ros-industrial/industrial_training #46: indigo build errors

jrgnicho, 2015-04-08:
The ccny_vision packages under the supplements directory will not build under ros-indigo.  Also, industrial_core is not a available as a debian release so it will have to be copied to the supplements or be manually installed in some way.

gavanderhoorn, 2015-04-08:
On 8-4-2015 20:44, Jorge Nicho wrote:
> The ccny_vision packages under the supplements directory will not build under ros-indigo.  Also, industrial_core is not a available as a debian release so it will have to be copied to the supplements or be manually installed in some way.  

fyi: industrial_core was recently released into Indigo, but the pkg
mirrors have not yet been synced completely.

Switching to ros-shadow-fixed should allow you to install it using apt-get.

shaun-edwards, 2015-04-08:
It's not installable under indigo.

shaun-edwards, 2015-04-08:
I mean it is NOW installable under indigo.

gavanderhoorn, 2015-04-08:
On 8-4-2015 22:22, Shaun Edwards wrote:
> I mean it is NOW installable under indigo.

you got me confused for a minute there :)

jrgnicho, 2015-04-08:
I just built PR #50 successfully with the debian install of industrial_core.

gomezc, 2015-04-09:
Merged the branch for this issue and confirmed successful build with changes to ccny_vision.

closed 2015-04-09 by gomezc