ros-industrial/industrial_training #56: Remove outdated universal_robot stack from the supplements

jrgnicho, 2015-04-16:
Remove the **universal_robot** stack and fix what breaks in the demo_manipulation or elsewhere in the training. Any package in the training that depends on **universal_robot** should build against the indigo debian install.

gomezc, 2015-04-17:
I think the cleanest way to deal with the supplements directory might be to delete the current contents and add back what we need, when we find that we need it. It seemed easier from a collaboration standpoint to delete the contents and then we each add the package that we need into the supplements as we go.

gavanderhoorn, 2015-04-17:
Out of curiosity: what is the rationale for adding copies of complete repositories to the training repository? Would it make sense to use a `.rosinstall` file for setting up the workspace? That would significantly reduce the maintenance overhead of the tutorials pkg.

If dependencies on particular versions is an issue: `.rosinstall` files can be setup to have `wstool` checkout specific tags or even commits.

jrgnicho, 2015-04-17:
@gavanderhoorn at the time, we wanted to lock on particular versions of such repos that worked with the training exercises without having to worry about updating them if changes in those repos were to break them.   Plus there were also a few local changes we made to those repos that were never made into a PR since they were only relevant to our training exercises.