ros-industrial/industrial_training #79: Descartes demo give inconsistent plans

shaun-edwards, 2015-10-02:
The Descartes demonstration give inconsistent plans between runs.  I've managed to track this down to inconsistent start/end positions calculated [here](https://github.com/ros-industrial/industrial_training/blob/indigo-devel/training/ref/demo_descartes/src/plan_and_run/src/tasks/plan_path.cpp#L31-L33).  It isn't clear whether this is a problem with the `GetClosestJointPoint` method itself or the underlying UR IK.

shaun-edwards, 2015-10-02:
@gavanderhoorn, @Jmeyer1292, this is the cause of odd behavior we saw.  A work-around is to comment out these lines and keep the first/last points as Cartesian points (i.e. don't change them).  This results in slower planning, but since @Jmeyer1292 improved Descartes performance, this isn't so much of an issue.

gavanderhoorn, 2015-10-02:
Yes, some colleagues of mine observed this as well, but assumed it was something they were doing wrong. Thanks for creating the issue.

gomezc, 2015-11-13:
@Jmeyer1292 did your PR address this issue? Can this issue be close?

Jmeyer1292, 2015-11-13:
@gomezc This PR addresses the "inconsistent" part of this issue. Consistently found plans may still have problems because of a related issue involving the seed state used to find the start/stop position. Not sure if that should be a separate issue.

gomezc, 2015-11-13:
@Jmeyer1292 I'll leave this open then, no need for a new issue.

shaun-edwards, 2015-11-13:
@Jmeyer1292, can you elaborate on the issue involving the seed state?

Jmeyer1292, 2015-11-13:
@shaun-edwards I actually withdraw that statement. The issue I noticed must have been in a modified version of this branch that I was sent by another user. The robot seed pose was always all-zeros instead of the actual robot position.

closed 2016-01-21 by shaun-edwards