ros-industrial/industrial_training #80: packages for session 1A and 2A

ghanimmukhtar, 2015-10-18:
Hi there, I have cloned the packages but the packages for the exercises 1A & 2A aren't included. Can anyone please guide me to where I can find them?
Thanks

gomezc, 2015-10-19:
I will update the wiki pages (http://aeswiki.datasys.swri.edu/rositraining/indigo/Exercises/). It is not at all clear, but the 1A and 2A sessions were part of an Advanced Track that we ran in March. We wanted to keep the advanced training material isolated from the basic/core training material.   
Exercise 1A utilizes the rosi_training package within the industrial_calibration_tutorials repository (https://github.com/ros-industrial/industrial_calibration_tutorials). Exercise 2A utilizes two packages within the industrial_pcl repository (https://github.com/ros-industrial/industrial_pcl). You will need to clone the respective repo in order to complete those exercises.

ghanimmukhtar, 2015-10-20:
@gomezc 
Thanks a lot, I will check it out and get back to you in case of further problems or doubts.

ghanimmukhtar, 2015-10-22:
@gomezc 
Never mind the last question, I was able to start it as a node (rosrun) the command is basically the same:
```
rosrun industrial_extrinsic_cal manual_calt_adjust my_camera_link .01 .0175
```
with the slight difference of how to pass the arguments.

gomezc, 2015-10-22:
@ghanimmukhtar  Thanks for pointing that out. So by changing "roslaunch" to "rosrun" you were able to get it to work? I will make that change on the wiki. Please let me know if you see any other typos/issues. This was the first time that we taught the Advanced material and there may be problems we didn't catch in class.

ghanimmukhtar, 2015-10-22:
@gomezc Yes,

ghanimmukhtar, 2015-10-22:
@gomezc  and of course if i saw something else i will let u know, thanks for this awesome tutorials

gomezc, 2015-11-12:
@ghanimmukhtar I'm going to close this issue as I updated the wiki with more info.

closed 2015-11-12 by gomezc