ros-industrial/industrial_training #84: Is tutorial 1A.2 for simulation or hardware manipulation?

ghanimmukhtar, 2015-11-18:
I have followed the instruction given in 1A.2, but at the end when I launch the "ur5_moveit_planning_execution.launch" it launches with some warnings and an error:
```
[WARN] [WallTime: 1447861877.268895] No calibration offset for joint "shoulder_pan_joint"
[WARN] [WallTime: 1447861877.270010] No calibration offset for joint "shoulder_lift_joint"
[WARN] [WallTime: 1447861877.271002] No calibration offset for joint "elbow_joint"
[WARN] [WallTime: 1447861877.271959] No calibration offset for joint "wrist_1_joint"
[WARN] [WallTime: 1447861877.272874] No calibration offset for joint "wrist_2_joint"
[WARN] [WallTime: 1447861877.273904] No calibration offset for joint "wrist_3_joint"
```

```
[ INFO] [1447861885.905860549]: MoveitSimpleControllerManager: Waiting for /follow_joint_trajectory to come up
[ INFO] [1447861890.906119520]: MoveitSimpleControllerManager: Waiting for /follow_joint_trajectory to come up
[ERROR] [1447861895.906415997]: MoveitSimpleControllerManager: Action client not connected: /follow_joint_trajectory
```

Furthermore in the script "ur5_moveit_planning_execution.launch" the simulation argument is set to false and an IP is given:
```
<!-- robot_ip: IP-address of the robot's socket-messaging server -->
  <arg name="sim" default="false" />
...........
  <arg name="robot_ip" default="192.168.1.13"/> 
```

So I am wondering is this package is intended for use with a hardware universal robot? And if so is there another script to do the exercise in simulation?

gomezc, 2015-11-24:
Hello @ghanimmukhtar . When we ran the training, we did it with real hardware (real cameras, target and UR robot). The tutorial is written around a real robot, programming it to move in the workspace, recording joint_states etc. 
There is currently not a good way to do this exercise in simulation, as the robot would be holding the target that the cameras are looking for. You can follow the tutorial at http://wiki.ros.org/industrial_extrinsic_cal/Tutorials/Multi_camera_calibration if you have cameras and a target (no robot). The setup files would be the same/similar but some of the other steps are different without a robot.

ghanimmukhtar, 2015-12-02:
thanks @gomezc  for your reply, so now I moved to the "manipulation demo", and when I tried to "catkin_make" I got an error in the node "move_to_wait_position.cpp", the error reads as follows:
```
/home/ghanim/industrial_training/training/work/demo_manipulation/src/collision_avoidance_pick_and_place/src/tasks/move_to_wait_position.cpp: In member function ‘void collision_avoidance_pick_and_place::PickAndPlace::move_to_wait_position()’:
/home/ghanim/industrial_training/training/work/demo_manipulation/src/collision_avoidance_pick_and_place/src/tasks/move_to_wait_position.cpp:27:64: error: cannot call member function ‘bool moveit::planning_interface::MoveGroup::setNamedTarget(const string&)’ without object
   moveit::planning_interface::MoveGroup::setNamedTarget ("wait");
```
So I looked in the code and changed line 27 from:
```
moveit::planning_interface::MoveGroup::setNamedTarget ("wait");
```
To:
```
move_group_ptr->setNamedTarget(cfg.WAIT_POSE_NAME);
```
And then rebuild the packages, and it worked without problems. My question is: since the method (setNamedTarget()) expect a reference as input why we pass a string (i.e. "wait")? Also I am wondering if what I did is okay or does it change something else?