ros-industrial/industrial_training #86: training_unit demo_manipulation fails to build

m-talha, 2016-02-28:
This exercise fails to build on my system due to the following error:

`/home/work/ros_industrial_training/training/work/demo_manipulation/src/robot_io/src/nodes/simulated_grasp_action_server.cpp:35:70: fatal error: object_manipulation_msgs/GraspHandPostureExecutionAction.h: No such file or directory #include <object_manipulation_msgs/GraspHandPostureExecutionAction.h>`

Copying the `object_manipulation_msgs` src folder from the supplements directory acts as a workaround.

gavanderhoorn, 2016-02-28:
See also [ROS-I training_unit demo_manipulation fails to build](http://answers.ros.org/question/227737/ros-i-training_unit-demo_manipulation-fails-to-build/) on ROS Answers.

haniehes, 2018-05-07:
I have the same issue and the above link from ROS answers couldn't help me. Is there any reliable solution for this issue?

jrgnicho, 2018-05-07:
On which branch are you seeing this issue?

haniehes, 2018-05-08:
It happened to me during building package dependencies (Demo 1.2), before launching MoveIT.  However, I have solved it by a trick. I built a temporary package in `perception_driven_ws` containing `GraspHandPostureExecutionAction.msg` and called this massage from there. It worked properly. Then I returned this message to `object_manipulation_msgs` and made my catkin.