ros-industrial/industrial_training #118: Error compiling exercise 4.1

gaetanauvray, 2017-04-27:
When trying to build this exercise, I get following error: 

"catkin_ws/src/ur5_demo_descartes/include/ur5_demo_descartes/ur_moveit_plugin.h:89:48: fatal error: moveit_msgs/GetKinematicSolverInfo.h: No such file or directory"

Reading this [post](https://github.com/ros-planning/moveit_msgs/issues/3), it seems that this GetKinematicSolverInfo service has been removed from package.

Any suggestion?

jrgnicho, 2017-04-27:
@gaetanauvray Which branch are you are you using?

gaetanauvray, 2017-04-27:
I am using kinetic for both, ros_industrial and movit_msgs.

gaetanauvray, 2017-04-28:
I was able to work around by putting the jade-devel branch of movit_msgs in my catkin environment, but solution might still be needed for kinetic branch.

guihomework, 2017-05-03:
Hi, had a similar issue, and fixed it in my IK plugin by this simple action

#include <moveit_msgs/GetKinematicSolverInfo.h>
should be replaced by 
#include <moveit_msgs/KinematicSolverInfo.h>

cl13554212322, 2017-06-05:
ros_kinetic fixed
1: add the moveit_ros_planning_interface in the find package() and catkin_package()
2:  add the depend in the package.xml:
    <build_depend>moveit_ros_planning_interface</build_depend>
    <run_depend>moveit_ros_planning_interface</run_depend>