ros-industrial/industrial_training #134: otion Planning CPP

cl13554212322, 2017-06-06:
Dear,
[ERROR] [1496711889.113917738]: RRTConnect: Unable to sample any valid states for goal tree
----------------------------------
[ INFO] [1496711882.678479230]: Loading robot model 'myworkcell'...
[ INFO] [1496711882.843591246]: Loading robot model 'myworkcell'...
[ INFO] [1496711884.005280932]: Ready to take commands for planning group manipulator.
ABORTED: No motion plan found. No execution attempted.

Please help me ,thank you !    ROS-Kinetic Version

ChenLong-Cristian, 2017-06-06:
I have finished all the tasks before the task "4.0 motion planning" and everything is ok, but I got stuck in here. please help me！

Levi-Armstrong, 2017-06-06:
Did you flip the target pose 180deg.

Levi-Armstrong, 2017-06-06:
If you did rotate the target pose 180 that is most likely the issue. It was left over and should of been removed. I have remove the statement from the wiki. Also, I tested the one in the exercise/4.0 and everything run. I would recommend comparing your code to the one that is in the exercise/4.0 directory.

Levi-Armstrong, 2017-06-16:
@ChenLong-Cristian Were you able to resolve the issue?

ChenLong-Cristian, 2017-06-18:
I am sorry for late response.
   I test the source code in 4.0 and 4.1, and copy relevant packages into the catkin_ws/src, but the same problem still exist. Have you tested the source code in a new computer for kinetic branch, not your own working computer, since I have to install some package for compailing in privious tutorials. I consider that my computer lack of some dependences. But the  catkin_make has no errro now. Moreover, I test the moveit tutorials for pr2 moveit_group_interface and everything is ok.
So,I still cannot find the problem in kinetic, but I have gone through all the tutorials in indigo branch and everything is ok,even if I have to install extra dependence package that haven't mentioned inthe tutorials. 
    Thanks for your research, tutorials and patient response. I will continue my study when I am free.

ChenLong-Cristian, 2017-08-09:
Now, all the tutorial is ok, thanks for the whole tutorials.

closed 2017-08-09 by ChenLong-Cristian