ros-industrial/industrial_training #147: Help making Descartes demo work with UR3?

JamesGiller, 2017-10-02:
Hi,

I have been following the [Descartes Planning and Execution](https://github.com/ros-industrial/industrial_training/wiki/Descartes-Planning-and-Execution----Application-Structure#2-list-all-the-packages-in-the-application) demo, which involves generating a path with Descartes and then executing it with a UR5 robot in RViz.

I have a real UR**3** robot, for which I want to write a program based on the demo; I have a couple of questions about how to accomplish this:

1) Is it necessary to create a corresponding "ur**3**_demo_descartes" package with "a custom Descartes Robot Model" for the UR**3**? I notice that in the demo [init_descartes.cpp](https://github.com/ros-industrial/industrial_training/blob/4a30b8f36c702ac0af0c6ca3e5349b7eea0862b6/exercises/Descartes_Planning_and_Execution/template_ws/src/plan_and_run/src/tasks/init_descartes.cpp#L21) we create an instance of [UR5RobotModel](https://github.com/ros-industrial/industrial_training/blob/4a30b8f36c702ac0af0c6ca3e5349b7eea0862b6/exercises/Descartes_Planning_and_Execution/template_ws/src/ur5_demo_descartes/include/ur5_demo_descartes/ur5_robot_model.h#L26), which derives from [URKinematicsPlugin](https://github.com/ros-industrial/industrial_training/blob/4a30b8f36c702ac0af0c6ca3e5349b7eea0862b6/exercises/Descartes_Planning_and_Execution/template_ws/src/ur5_demo_descartes/include/ur5_demo_descartes/ur_moveit_plugin.h#L111); however, in the [kinematics.yaml](https://github.com/ros-industrial/industrial_training/blob/kinetic-devel/exercises/Descartes_Planning_and_Execution/template_ws/src/ur5_demo_moveit_config/config/kinematics.yaml) we specify a KDLKinematicsPlugin. Does the "Inverse-Kinematics closed form solution" from the URKinematicsPlugin even get used? Or does the program use KDL?

2) If it is necessary to create a "ur**3**_demo_descartes" package, can I just base it off of the files in the [ur_kinematics](https://github.com/ros-industrial/universal_robot/tree/kinetic-devel/ur_kinematics) package from the universal_robot stack? It seems that ur_kin.h/cpp & ur_moveit_plugin.h/cpp are almost a copy paste; however [ur5_robot_model](https://github.com/ros-industrial/industrial_training/blob/kinetic-devel/exercises/Descartes_Planning_and_Execution/template_ws/src/ur5_demo_descartes/src/ur5_robot_model.cpp) has source specific to "ur5_demo_descartes". Is there any guidance on how to create a corresponding "ur**3**_robot_model" source?

This training material is very good overall, but I think learners would really benefit from having more information about **why** different files exist and **how** they connect to achieve the desired result. As it stands, I think that most people would not be able to deviate too much from the specific examples and demos due to not knowing how the pieces (e.g. Kinematics Plugins, Descartes Robot Models) are created and combined.

demorise22ng, 2018-02-07:
I have the same challenge. I have a custom robot and I'm wondering how to create a robot_model for it. It looks like the robot_model package contains all the possible IK and FK solutions of the robot, which descartes uses for planning. Any help on how to create this robot_model package will be highly appreciated.

jrgnicho, 2018-02-07:
@demorise and @JamesGiller sorry I just caught up with this issue, hopefully this reply is still relevant.  When we first wrote that tutorial there were limitations with the moveit kinematics base class that prevented us from getting IK multiple solutions for a given tool position.  This step is critical for Descartes so we created the Descartes Robot Model in part as a workaround to that limitation.  The moveit KinematicsBase has since then been modified and it is no longer necessary to create a Descartes Robot Model package specific to your robot.   Instead you can instantiate an [IkFastMoveitStateAdapter](https://github.com/ros-industrial-consortium/descartes/blob/kinetic-devel/descartes_moveit/include/descartes_moveit/ikfast_moveit_state_adapter.h)  object which will load the kinematics plugin of your arm as long as it is provided in the "kinematics.yaml" of the moveit config package.  This flavor of the Descartes Robot Model assumes the the underlying kinematics is running a closed-form solver  and so it'll exploit that fact in order to generate more pose samples and potentially produce a better path.  
In short you shouldn't have to create a  Descartes Robot package for your robot in kinetic, just replace this [line](https://github.com/ros-industrial/industrial_training/blob/kinetic/exercises/Descartes_Planning_and_Execution/solution_ws/src/plan_and_run/src/tasks/init_descartes.cpp#L21) to create an instance of IkFastMoveitStateAdapter.  Hope this helps

@Jmeyer1292 feel free to chime in on this.

demorise, 2018-02-07:
@jrgnicho Thanks a lot. That's really helpful.

closed 2018-07-06 by jrgnicho