ros-industrial/industrial_training #151: Suggestion: Add code coverage analysis to unit testing section

fkromer, 2017-10-12:
I would suggest to add an "how to" section about adding *code coverage analysis* (e.g. using the [code coverage cmake module](https://github.com/bilke/cmake-modules/blob/master/CodeCoverage.cmake)) to the [unit testing](http://ros-industrial.github.io/industrial_training/_source/session6/Unit-Testing.html#) section.

UniverseHan, 2020-06-17:
any progress about this?

fkromer, 2020-06-18:
@UniverseHan Nope

marip8, 2020-06-18:
It's not strictly a tutorial, but you can check out the initial PR for the [cmake_common_scripts project](https://github.com/ros-industrial/cmake_common_scripts/pull/1/files). @Levi-Armstrong has started curating some commonly used CMake macros (including code coverage) into this project for easier use by other projects. There is an example in the README that shows how to add code coverage to a target

UniverseHan, 2020-09-15:
@fkromer Ok, I am happy to contribute to this subject. let me try.

fkromer, 2020-09-21:
@UniverseHan Cool. Thx.