ros-industrial/industrial_training #156: Respect to:  industrial_training/gh_pages/_source/demo1/Package-Setup.md

surfertas, 2017-10-23:
Hello I havent had any troubles following the training thus far, but have just encountered an error when attempting to build the packages found in `perception_driven_ws` by running the below:

`catkin build --cmake-args -G 'CodeBlocks - Unix Makefiles'`

```
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by
  "industrial_robot_client" with any of the following names:

    industrial_robot_clientConfig.cmake
    industrial_robot_client-config.cmake

  Add the installation prefix of "industrial_robot_client" to
  CMAKE_PREFIX_PATH or set "industrial_robot_client_DIR" to a directory
  containing one of the above files.  If "industrial_robot_client" provides a
  separate development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:4 (find_package)
```

jrgnicho, 2017-11-07:
Can you provide a link to the instructions you followed?

jrgnicho, 2018-07-06:
The instruction to run  catkin build with the "CodeBlocks" option 
```
catkin build --cmake-args -G 'CodeBlocks - Unix Makefiles
```
Is no longer required.  Instances of this command should be updated to ```catkin build``` only