ros-industrial/industrial_training #159: Demo 1.2 not working (Error on catkin build)

StefH4711, 2017-11-10:
Hi guys,

I´m new to ROS and learned many things following the ROS-Industrial tutorial. Till now all lessons worked well but not the Demo 1.2. I followed all steps described here and inside of the chapters before, but I am always getting an error in catkin build.

See the output of catkin build below. Any help would be appreciated:

`she@ubuntu:~/perception_driven_ws$ catkin build --cmake-args -G 'CodeBlocks - Unix Makefiles'
-----------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /home/she/catkin_ws/devel:/opt/ros/kinetic
Workspace:                   /home/she/perception_driven_ws
-----------------------------------------------------------------------
Source Space:       [exists] /home/she/perception_driven_ws/src
Log Space:          [exists] /home/she/perception_driven_ws/logs
Build Space:        [exists] /home/she/perception_driven_ws/build
Devel Space:        [exists] /home/she/perception_driven_ws/devel
Install Space:      [unused] /home/she/perception_driven_ws/install
DESTDIR:            [unused] None
-----------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-----------------------------------------------------------------------
Additional CMake Args:       -G CodeBlocks - Unix Makefiles
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-----------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-----------------------------------------------------------------------
Workspace configuration appears valid.
-----------------------------------------------------------------------
[build] Found '34' packages in 0.0 seconds.                                                                                                                                        
[build] Package table is up to date.                                                                                                                                               
Starting  >>> ar_track_alvar_msgs                                                                                                                                                  
Finished  <<< ar_track_alvar_msgs                                  [ 0.9 seconds ]                                                                                                 
Starting  >>> controller_manager_msgs                                                                                                                                              
Finished  <<< controller_manager_msgs                              [ 1.2 seconds ]                                                                                                 
Starting  >>> hardware_interface                                                                                                                                                   
Finished  <<< hardware_interface                                   [ 0.2 seconds ]                                                                                                 
Starting  >>> household_objects_database_msgs                                                                                                                                      
Finished  <<< household_objects_database_msgs                      [ 1.6 seconds ]                                                                                                 
Starting  >>> realtime_tools                                                                                                                                                       
Finished  <<< realtime_tools                                       [ 0.3 seconds ]                                                                                                 
Starting  >>> robot_config                                                                                                                                                         
Finished  <<< robot_config                                         [ 0.2 seconds ]                                                                                                 
Starting  >>> sensor_config                                                                                                                                                        
Finished  <<< sensor_config                                        [ 0.2 seconds ]                                                                                                 
Starting  >>> ur_description                                                                                                                                                       
Finished  <<< ur_description                                       [ 0.2 seconds ]                                                                                                 
Starting  >>> ur_kinematics                                                                                                                                                        
Finished  <<< ur_kinematics                                        [ 0.5 seconds ]                                                                                                 
Starting  >>> ur_msgs                                                                                                                                                              
Finished  <<< ur_msgs                                              [ 1.0 seconds ]                                                                                                 
Starting  >>> ar_track_alvar                                                                                                                                                       
___________________________________________________________________________________________________________________________________________________________________________________
Errors     << ar_track_alvar:make /home/she/perception_driven_ws/logs/ar_track_alvar/build.make.010.log                                                                            
c++: internal compiler error: Getötet (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
make[2]: *** [CMakeFiles/ar_track_alvar.dir/src/Marker.cpp.o] Fehler 4
make[1]: *** [CMakeFiles/ar_track_alvar.dir/all] Fehler 2
make: *** [all] Fehler 2
cd /home/she/perception_driven_ws/build/ar_track_alvar; catkin build --get-env ar_track_alvar | catkin env -si  /usr/bin/make --jobserver-fds=8,9 -j; cd -
...................................................................................................................................................................................
Failed     << ar_track_alvar:make                                  [ Exited with code 2 ]                                                                                          
Failed    <<< ar_track_alvar                                       [ 19.6 seconds ]                                                                                                
Abandoned <<< combined_robot_hw                                    [ Unrelated job failed ]                                                                                        
Abandoned <<< controller_interface                                 [ Unrelated job failed ]                                                                                        
Abandoned <<< joint_limits_interface                               [ Unrelated job failed ]                                                                                        
Abandoned <<< transmission_interface                               [ Unrelated job failed ]                                                                                        
Abandoned <<< manipulation_msgs                                    [ Unrelated job failed ]                                                                                        
Abandoned <<< ur5_collision_avoidance_moveit_config                [ Unrelated job failed ]                                                                                        
[... cut ...]
[build]   Warnings:  None.                                                                                                                                                         
[build]   Abandoned: 20 packages were abandoned.                                                                                                                                   
[build]   Failed:    1 packages failed.                                                                                                                                            
[build] Runtime: 26.6 seconds total. `

gavanderhoorn, 2017-11-10:
> `c++: internal compiler error: Getötet (program cc1plus)`

I typically see these kinds of errors on platforms that are memory constrained. C++ code - especially heavily templated code - can need  a lot of memory to compile.

Are you doing this on a RPi or another SBC?

In a VM perhaps?

StefH4711, 2017-11-13:
That was it. Extending the memory of my VM from 1 GB to 4 GB solved the issue. Thanks for your help.

closed 2017-11-13 by StefH4711