ros-industrial/industrial_training #160: Create ROS 2 Training Material

Levi-Armstrong, 2017-11-14:
This issues is to open discussion related to creating ROS 2 training material.

Briancbn, 2020-05-05:
@Levi-Armstrong from ROS-I AP. It seems that there aren't a lot of discussion or activities here at the moment. I have several questions and suggestions regarding the ROS 2 training material.
- **Eloquent** looks good for the development at the moment, but will EOL this Nov. A lot of the **eloquent** features and interfaces like **launch_xml**, **rviz markers**, etc. are not available in **Dashing**. **Foxy** should be the better choice once it's out, hopefully mid June.
- Basic training materials from 1.0 - 3.2, 6.1, 6.2, seem ready for porting. 
- The others related to **Moveit** needs to wait for **Moveit2** to be ready. The current **Moveit2 Beta** doesn't have `move_group_interface` yet, although mostly functional, so I think these should be after **Moveit2 Foxy**.
- I am not clear on the plan for **Descartes** yet. Will it be port to ROS2 or it will be part of **Moveit2** future milestone in its [Development Roadmap](https://moveit.ros.org/documentation/contributing/roadmap/).
- A list ROS2 features that needs tutorials here, like **lifecycle_node**, **qos**, **composition** etc..
- Can we also track the porting process either here or somewhere else?

cardboardcode, 2021-04-23:
As of this writing, here is an updated response to the questions @Briancbn have, 

## Questions

> * **Eloquent** looks good for the development at the moment, but will EOL this Nov. A lot of the **eloquent** features and interfaces like **launch_xml**, **rviz markers**, etc. are not available in [...]

## Responses

1. There is now a ROS2 Foxy port of the documentation. Please access it via [this link](https://industrial-training-master.readthedocs.io/en/foxy/).

2. Based on the current ROS2 Foxy port, here is quick overview of the exercises/sections that was mentioned in the question.
**Exercise 1.0, ROS Setup** - Done. 
**Exercise 1.1, Creating Workspace** - Done.
**Exercise 1.2, Installing Packages** - Done.
**Exercise 1.3, Packages and Nodes** - Done.
**Exercise 1.4, Topics and Messages** - Done.
**Exercise 2.0, Services** - Done.
**Exercise 2.1, Actions** - Outsourced to ROS2 general tutorials.
**Exercise 2.2, Launch Files** - Done.
 **Exercise 2.3, Parameters** - Done.
**Exercise 3.0, Intro to URDF** - Done.
**Exercise 3.1, Workcell XACRO** - Done.
**Exercise 3.2, Services** - Done.
**Exercise 6.1, Unit Testing** - Incomplete. Still uses catkin.
**Exercise 6.2, Using rqt toosl for analysis** - Incomplete. Still uses ROS1 roslaunch.

3. MoveIt2 has been ported to ROS2 Foxy. Please access it via [this link](https://moveit.ros.org/). It now has `move_group_interface`.

4. As of this writing, **DeCartes**-related exercises in the Training Material have not been properly ported with them still using catkin. There are also no ongoing online discussions that I can find on porting Decartes to ROS2 as well. 

5. Training Materials for `lifecycle_node`, `QOS` and `composition` have not been created. For now, ROS2 users can learn more about them through the general ROS2 docs. Access them via the links below.
 
**[lifecycle_node](https://github.com/ros2/demos/blob/foxy/lifecycle/README.rst)**
**[QOS](https://docs.ros.org/en/foxy/Tutorials/Quality-of-Service.html)**
**[composition](https://docs.ros.org/en/foxy/Tutorials/Composition.html)**

6. @Levi-Armstrong At the moment, with your permission, **can we track the porting process here**? Do let us know if there are other preferred threads ROS-Industrial Consortium America already has so as not to duplicate porting efforts.

cardboardcode, 2021-04-23:
@jdlangs @JeremyZoss 
Thank you for ROS2 porting efforts thus far.

Do let us know the current state of the ROS2 porting here. Our team at **ROS-Industrial Consortium Asia Pacific** would like to help out as well.