ros-industrial/industrial_training #163: Minor Issue: Links to wiki have been updated

JP-SG-ROS, 2017-11-18:
Relatively minor issue with a quick fix. There are a number of hyperlinks from the training material that seem to reference the old wiki page. These can easily be updated by replacing references of

`
http://www.ros.org/wiki/ROS/Tutorials/_package_name_
`
to:
`
http://wiki.ros.org/ROS/Tutorials/_package_name_
`
eg 

http://www.ros.org/wiki/ROS/Tutorials/BuildingPackages
should be changed to:
http://wiki.ros.org/ROS/Tutorials/BuildingPackages

etc.
Thanks for the great materials!

jdlangs, 2017-12-27:
Thanks @JP-SG-ROS! Would you be able to quickly put together a PR to make those changes?

JP-SG-ROS, 2017-12-27:
Might not be neccessary now? (they have added an automatic re-direct from the old wiki page to the new one)

gavanderhoorn, 2018-02-13:
The redirect is known to 'sometimes' stop working.

I recommend search-replacing all occurences of the old url with the new one to avoid potential future stale links.

closed 2018-07-06 by jrgnicho