ros-industrial/industrial_training #170: 4.2 Intro to Perception broken file link table.pcd

AdamSorrel, 2018-01-08:
The link in the section "Kinect Sensor Example" prompting to download file _PointCloud file_ is not working. The proper link should be probably pointing towards [this file](https://github.com/ros-industrial/industrial_training/blob/kinetic/gh_pages/_downloads/table.pcd). 

I apologize is this is this is too petty, but some of us are not as sharp. I for once have downloaded a web browser error message and then I was confused why it wouldn't load. 

Thanks a lot for great tutorials!

Adam

jrgnicho, 2018-02-07:
@AdamSorrel thanks for pointing this out.  We'll address it shortly.

jdlangs, 2018-02-07:
This was actually long-ago fixed by #155. @Levi-Armstrong, is there Sphinx auto build-and-deploy mechanism that's not working or do the generated pages just need a manually update?

jrgnicho, 2018-07-06:
Fixed by #196

closed 2018-07-06 by jrgnicho