ros-industrial/industrial_training #174: Your exercises webpage cannot be opened

haniehes, 2018-01-29:
I am not sure what is the reason, but I cannot open ROS Industrial (Kinetic) Training Exercises page. As before, I click on link blew:
ros-industrial.github.io/industrial_training/
but it is just loading. This problem belongs to kinetic exercises.

jrgnicho, 2018-01-29:
It's working for me at the moment.  Could you try again?

haniehes, 2018-01-30:
I think the website is so busy! I could open it suddenly during the last night (11 pm) but it does not work during the working hours!

gavanderhoorn, 2018-01-30:
It works for me as well - and during working hours.

Could there be some local disturbance / something blocking you from accessing github?

haniehes, 2018-01-30:
No, I have access to other `github.io` and the infinite loading appears only on this website. I have encountered this problem for more than 2 weeks.

gavanderhoorn, 2018-01-30:
Taking a look with the Firefox performance and network monitor I don't see anything that stands out. Most assets load within 50ms for me.

Can you try clearing your browser cache, then open the site again?

And as inconvenient as it is, this is most likely not a problem with the `industrial_training` repository, but with something between you and where it's hosted (unless there is some resource that is retrieved from a CDN or something, like google analytics).

gavanderhoorn, 2018-01-30:
@haniehes wrote:
> I could open it suddenly during the last night (11 pm) 

was this at home?

> but it does not work during the working hours!

and is this 'at work' (ie: a different internet connection)?

haniehes, 2018-01-30:
Yes for this case. But it also happened during the weekend, when I was at home (different internet connection and Firefox as a browser). Also, I have seen this problem on my colleague's computers. So, as it is not related to `industrial_training` website, I can continue with `Github` version.

gavanderhoorn, 2018-01-30:
If you are using firefox now, could you try and use the web developer tools (integrated these days) and check the *performance* and *network* tabs while loading `http://ros-industrial.github.io/industrial_training`?

That should give us an idea of what is going on, if we're lucky.

> So, as it is not related to `industrial_training` website, I can continue with `Github` version. 

I meant that unless this is an issue with something (like a script, or an asset) that is loaded from a CDN other than Github's that was added by @Levi-Armstrong (google analytics, something else), this doesn't seem to be an issue directly with the site in this repository, but perhaps a more general issue with *something* between you and `http://ros-industrial.github.io/industrial_training`.

haniehes, 2018-01-30:
I opened these tabs, but what should I look for? The network status is a gray point and not connected.
Also, I tested it with my phone connection and I can open it. I am sure that it depends on my workplace connections and sometimes my home(!).

gavanderhoorn, 2018-01-30:
Make sure the *network* tab is active, tick the `Disable cache` checkbox then try to open `http://ros-industrial.github.io/industrial_training` again (use the same tab, not a new one).

>I am sure that it depends on my workplace connections and sometimes my home(!).

it would seem that way yes.

In that case I feel that this issue tracker is not the best place to diagnose this, but a more generic forum.

haniehes, 2018-01-30:
Thanks for your follow up. I have done these procedures, but still gray circle for status. 
So, I have to find a better internet connection and close this issue.

gavanderhoorn, 2018-01-30:
I don't understand what you mean with "gray circle for status". The 'network' tab should at least show some activity (like an outgoing request).

This is what I get when I do a test with the tutorial page:

[image]

haniehes, 2018-01-30:
and that is what I recorded:
[image]

[image]

gavanderhoorn, 2018-01-30:
Does `nc ros-industrial.github.io 80` work for you? Try typing `get /` after it connects.

Does `wget http://ros-industrial.github.io/industrial_training` work?

haniehes, 2018-01-30:
You mean that I have to include 'nc' and '80' to the main link in the address bar?
If it is, non of them works.

gavanderhoorn, 2018-01-30:
No. Both of those lines are command lines that you should enter in a `bash` session / terminal session.

haniehes, 2018-01-30:
No. They don't make any changes.

gavanderhoorn, 2018-01-30:
If the site does open on your phone as you mention, I believe it would seem there is either a (corporate) proxy or something like a firewall between you and those hosts that is preventing you from connecting to them.

There's nothing more I can do from here unfortunately.

haniehes, 2018-01-30:
Thanks for your prompt responses. Yes, I understand and I will continue through Github.

closed 2018-01-30 by haniehes

gavanderhoorn, 2018-01-30:
A final question:

@haniehes wrote:
> @gavanderhoorn wrote:
> > No. Both of those lines are command lines that you should enter in a `bash` session / terminal session.
> 
> No. They don't make any changes.

if with "they don't make any changes" you mean that Firefox / your browser still can't open the site, then that is expected. Those commands were not meant to fix that.

The output of `wget` should provide some more information on what is going on. Did `wget` not print any output after you executed it?

haniehes, 2018-01-30:
I didn't receive any reaction in terminal or somewhere else from the first command and the second one had this result in terminal:
`--2018-01-30 16:17:14--  http://ros-industrial.github.io/industrial_training
Resolving ros-industrial.github.io (ros-industrial.github.io)... 151.101.37.147, 2a04:4e42:9::403
Connecting to ros-industrial.github.io (ros-industrial.github.io)|151.101.37.147|:80... `

gavanderhoorn, 2018-01-30:
Ok. That does look like what I wrote earlier: something (or someone) is blocking you from reaching that host.