ros-industrial/industrial_training #175: Links to source code (on GH) do not appear to use permanent urls

gavanderhoorn, 2018-02-13:
Example: [this line](https://github.com/ros-industrial/industrial_training/blame/fd627e537ae86a244d3bca4010ecaf4230145be5/gh_pages/_source/session3/Intro-to-URDF.md#L41) in `Intro to URDF` links to `github.com/ros-industrial/abb/blob/indigo-devel/abb_irb2400_support/urdf/irb2400_macro.xacro` lines 56 to 71.

From the highlight on the page (after clicking on the link) it would appear that the file (`irb2400_macro.xacro`) has already been changed since the author linked to it, as the highlight seems to be shifted down by 2 lines from where it originally was.

Links to source files in GH repositories should use parmanent urls (ie: urls that link to specific versions), instead of linking to branches. That way we avoid files in repositories changing and links in documentation like `industrial_training` from going stale.

See [Getting permanent links to files](https://help.github.com/articles/getting-permanent-links-to-files) in the GH documentation for how to retrieve permanent urls for any file (or directory) in a gh repository (essentially: pressing `y` an copying the new url).

gavanderhoorn, 2018-02-13:
Example permanent url: `https://github.com/ros-industrial/abb/blob/84825661073a18e33b68bb01b5bf371edd2efd49/abb_irb2400_support/urdf/irb2400_macro.xacro#L54-L69`.

jrgnicho, 2018-02-13:
I agree with using permanent links, I didn't know of this feature.

gavanderhoorn, 2018-02-13:
Note that the URLs are 'permanent' because they include a commit hash. As those are not supposed to change - unless the history of the repository is rewritten - the URL is essentially permanent.

jrgnicho, 2018-02-13:
Right

HsGaurav, 2018-07-11:
The commit permanent link - https://github.com/ros-industrial/abb/blob/84825661073a18e33b68bb01b5bf371edd2efd49/abb_irb2400_support/urdf/irb2400_macro.xacro#L56-L71
Is this type of permanent link that was to be made?

PR #210 (2018-07-16): Update link specific link to be a permalink and fix line references.
Fixes #175
commit a1de425c (2018-07-16): Update link specific link to be a permalink and fix line references.
Fixes #175
closed 2018-07-16 by Levi-Armstrong