ros-industrial/industrial_training #202: Demo 1 cannot be run on real robot

haniehes, 2018-06-01:
I tried to run Demo.1 on my UR5 through Ros kinetic. The simulation of Pick and Place movement is done completely. Also, I see the real robot I but when I try to run the simulated perception on the real robot I have this two errors:
`Error:   manipulator[RRTConnectkConfigDefault]: Motion planning start tree could not be initialized!
         at line 172 in /tmp/binarydeb/ros-kinetic-ompl-1.2.3/src/ompl/geometric/planners/rrt/src/RRTConnect.cpp`
Hence, no goal sampling is done and I cannot move my robot.
Previously I faced another error [#200](https://github.com/ros-industrial/industrial_training/issues/200) but I fixed it by changing robot connection.
 I would appreciate if anybody could help me in this regard.

jrgnicho, 2018-07-06:
You may want to make sure that the start pose of the real robot isn't in violation of the software (urdf) joint limits.

closed 2018-07-11 by haniehes

chcorbato, 2018-07-11:
@haniehes is the issue solved? Please kindly indicate how it was solved and if any further action is required.

Thanks!

haniehes, 2018-07-11:
Yes, my issue solved. The problem was planning to reach the Wait position. At first, my Elbow angle was between -360 and 0. When I changed it to 0 to 360, the robot could start motion planning.
Thanks