ros-industrial/industrial_training #220: Replace `xacro.py` to `xacro --inorder`

youtalk, 2018-08-29:
According to [ROS wiki](http://wiki.ros.org/xacro#Processing_Order), calling `xacro.py` is deprecated since ROS Jade. We should call `xacro --inorder` instead.

For example, when I launch the following `roslaunch`, I found the warning.

```
$ roslaunch myworkcell_support view_urdf.launch 
... logging to /home/youtalk/.ros/log/28629fea-ab96-11e8-970b-001c4274c9d5/roslaunch-{$HOST}-13712.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
xacro.py is deprecated; please use xacro instead
...
```

There are over 50 search results which uses `xacro.py` in `ros-industrial` packages.
https://github.com/search?p=2&q=org%3Aros-industrial+xacro.py&type=Code

So I started replacing all of them.

gbiggs, 2018-08-30:
From Melodic, `--inorder` is no longer needed (it has been the default for a while). If it is used, a warning will be printed.

    $ xacro --inorder
    xacro: in-order processing became default in ROS Melodic. You can drop the option.
    Usage: xacro [options] <input>

youtalk, 2018-08-30:
Oops. Thank you for your comment.
Because I've just used ROS Kinetic, I couldn't notice it.
I will update my PRs.

gavanderhoorn, 2018-08-30:
@youtalk: what @gbiggs writes is correct, but since we don't target Melodic yet with most ROS-Industrial packages, adding `--inorder` is still ok. I don't believe you have to update your PRs.

youtalk, 2018-08-30:
Oh, I got it.

youtalk, 2018-08-31:
I've finished replacing. I hope someone reviews them.

closed 2018-09-18 by Levi-Armstrong

isaacKL, 2020-02-05:
So how should that look in line?

darshb34, 2020-02-18:
> So how should that look in line?
                For ROS Melodic
<param name="robot_description" command="$(find xacro)/xacro.py $(arg model)" />

                For ROS kinetic and others
<param name="robot_description" command="$(find xacro)/xacro --inorder $(arg model)" />

Msamaritan, 2021-02-11:
xacro: in-order processing became default in ROS Melodic. You can drop the option.
substitution args not supported:  No module named 'rospkg'
when processing file: /home/cyber-mysterio/catkin_ws/src/turtlebot3/turtlebot3_description/urdf/turtlebot3_burger.urdf.xacro
RLException: Invalid <param> tag: Cannot load command parameter [robot_description]: command [['/opt/ros/noetic/lib/xacro/xacro', '--inorder', '/home/cyber-mysterio/catkin_ws/src/turtlebot3/turtlebot3_description/urdf/turtlebot3_burger.urdf.xacro']] returned with code [2]. 

Param xml is <param name="robot_description" command="$(find xacro)/xacro --inorder $(find turtlebot3_description)/urdf/turtlebot3_$(arg model).urdf.xacro"/>
The traceback for the exception was written to the log file

I'm in ROS Noetic What should I do now? Anyone!?

gavanderhoorn, 2021-02-11:
I'm not sure, but it seems you're not using any of the packages in this repository.

Could I ask you to post your problem on a more relevant forum?