ros-industrial/industrial_training #238: Exercise 2.3- Parameter, base_frame not declared

tabelh, 2019-01-17:
Hi I am pretty new to ROS and the programming in c++ in general. I've been following the ROS-Industrial training material, but have been facing an issue for a while now. When I tried to make the package i got the following error: 
/home/thomas/catkin_ws/src/myworkcell_core/src/myworkcell_node.cpp:19:30: error: ‘base_frame’ was not declared in this scope
     srv.request.base_frame = base_frame;
                              ^
/home/thomas/catkin_ws/src/myworkcell_core/src/myworkcell_node.cpp: In function ‘int main(int, char**)’:
/home/thomas/catkin_ws/src/myworkcell_core/src/myworkcell_node.cpp:46:56: error: ‘base_frame’ was not declared in this scope
   private_node_handle.param<std::string>("base_frame", base_frame, "world"); // parameter name, string ob
       
The previous exercise compiled without an issue.                                               
As i understand it i have to define base_frame but as i understand it from the exercise this should be done using the private_node_handle. 

I added all relevant programs, i think.. I did not include the fake_ar_publisher as I did not alter any of the code and therefore doubt it could be what's the cause.

If I anyone could help me that would be very much appreciated. 

launch file
[workcelllaunch.txt](https://github.com/ros-industrial/industrial_training/files/2769430/workcelllaunch.txt)
service file
[LocalizePart.txt](https://github.com/ros-industrial/industrial_training/files/2769421/LocalizePart.txt)
node
[myworkcell_node.txt](https://github.com/ros-industrial/industrial_training/files/2769422/myworkcell_node.txt)
node
[vision_node.txt](https://github.com/ros-industrial/industrial_training/files/2769427/vision_node.txt)

marip8, 2019-01-17:
@tabelh there are two issues in your `myworkcell_node.cpp` file, and both result from trying to use the variable `base_frame` which hasn't been declared yet. In C++, unlike some other languages such as Python, all variables have to be declared before you can use them

1. The function signature of the `start` method changes in 2.3 Scan-N-Plan Application: Guidance step 4:
    - `void start()` -> `void start(const std::string& base_frame)`

    In your `start` method, no variable called `base_frame` has been created, but if you change the 
    function signature as described, that variable is declared as an argument to the function
1. In the `main` function you've tried to use a variable called `base_frame` that hasn't been defined yet in the function. To fix this, just declare the `base_frame` variable above the parameter call:
    ```
    std::string base_frame;
    private_node_handle.param<std::string>("base_frame", base_frame, "world");
    ```
    The node handle's `param` method will then do the work of assigning the value of the parameter to the `base_frame` string

tabelh, 2019-01-18:
Thank you very much good Sir.

closed 2019-02-26 by marip8