ros-industrial/industrial_training #239: Exercise 1.2 Catkin build fail

HareshMiriyala, 2019-01-22:
cd ~/catkin_ws/src
git clone https://github.com/jmeyer1292/fake_ar_publisher.git

I cloned the fake_ar_publisher and did a catkin build, but the build failed citing the following errors : 

Errors     << fake_ar_publisher:make /home/haresh/catkin_ws/logs/fake_ar_publisher/build.make.003.log             
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/gencpp/cmake/../../../lib/gencpp/gen_cpp.py", line 49, in <module>
    srv_template_map)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genmsg/template_tools.py", line 213, in generate_from_command_line_options
    generate_from_file(argv[1], options.package, options.outdir, options.emdir, options.includepath, msg_template_dict, srv_template_dict)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genmsg/template_tools.py", line 154, in generate_from_file
    _generate_msg_from_file(input_file, output_dir, template_dir, search_path, package_name, msg_template_dict)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genmsg/template_tools.py", line 99, in _generate_msg_from_file
    search_path)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genmsg/template_tools.py", line 77, in _generate_from_spec
    interpreter = em.Interpreter(output=ofile, globals=g, options={em.RAW_OPT:True,em.BUFFERED_OPT:True})
AttributeError: module 'em' has no attribute 'Interpreter'
make[2]: *** [/home/haresh/catkin_ws/devel/.private/fake_ar_publisher/include/fake_ar_publisher/ARMarker.h] Error 1
make[2]: *** Deleting file '/home/haresh/catkin_ws/devel/.private/fake_ar_publisher/include/fake_ar_publisher/ARMarker.h'
make[1]: *** [CMakeFiles/fake_ar_publisher_generate_messages_cpp.dir/all] Error 2
make: *** [all] Error 2
cd /home/haresh/catkin_ws/build/fake_ar_publisher; catkin build --get-env fake_ar_publisher | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -

hipships, 2019-05-02:
do "catkin_make" instead "cakin build"

schornakj, 2019-06-13:
There have been reports of a namespace collision between `empy`, a core ROS Python dependency, and `em`, a non-ROS terminal tool. Check to see that you only have `empy` installed and not `em`.

Related issues:
https://github.com/ros/genmsg/issues/63
https://github.com/ros-drivers/rosserial/issues/204

In general I'd suggest sticking with `catkin build` instead of  reverting to `catkin_make`. They do the same thing under the hood and it's a bit challenging to switch from one to the other after you've built your workspace.

closed 2019-06-14 by HareshMiriyala