ros-industrial/industrial_training #267: [Demo 1] hw_interface argument to ur_arm_transmission xacro macro

jdlangs, 2019-06-17:
The current UR5 description package requires a `hw_interface` argument for the `ur_arm_transmission` xacro macro which the copy of `ur5.urdf.xacro` in the training material [is missing](https://github.com/ros-industrial/industrial_training/blob/245438735fb353d8e027378622278bb791d85e9b/exercises/Perception-Driven_Manipulation/template_ws/src/robot_config/urdf/ur5.urdf.xacro#L313). After a brief comparison I didn't see any notable difference between this file and the one from `ur_description` so probably this file should just be deleted and the UR5 macro from the official package should be used instead.

ngwk1, 2019-07-02:
#272 Added additional parameters `hw_interface` to overcome compilation error. Please review the changes.

marip8, 2020-07-09:
Closing; addressed by #315

closed 2020-07-09 by marip8