ros-industrial/industrial_training #268: eigen error in demo_application.cpp

IsaacPiersall, 2019-06-18:
I get the following error, it seems to have something to do with the eigen matrix initialization in demo_application.cpp.
 
**************console output**********************

roslaunch plan_and_run demo_run.launch 
... logging to /home/ros5/.ros/log/8ab3e7f0-91df-11e9-a030-d43b04c1c3b6/roslaunch-ROS5-20046.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ROS5:40137/

SUMMARY
========

PARAMETERS
 * /plan_and_run_node/base_link: base_link
 * /plan_and_run_node/group_name: manipulator
 * /plan_and_run_node/tip_link: tool
 * /plan_and_run_node/trajectory/center: [0.36, 0.2, 0.1]
 * /plan_and_run_node/trajectory/foci_distance: 0.07
 * /plan_and_run_node/trajectory/num_lemniscates: 4
 * /plan_and_run_node/trajectory/num_points: 200
 * /plan_and_run_node/trajectory/radius: 0.08
 * /plan_and_run_node/trajectory/seed_pose: [0.0, -1.03, 1.57...
 * /plan_and_run_node/trajectory/time_delay: 0.1
 * /plan_and_run_node/visualization/min_point_distance: 0.02
 * /plan_and_run_node/world_frame: world
 * /rosdistro: melodic
 * /rosversion: 1.14.3

NODES
  /
    plan_and_run_node (plan_and_run/plan_and_run_node)

ROS_MASTER_URI=http://localhost:11311

process[plan_and_run_node-1]: started with pid [20061]
[ INFO] [1560872557.493717105]: Loaded application parameters
[ INFO] [1560872557.494220388]: Task 'loadParameters' completed
[ INFO] [1560872557.832014921]: Connected to 'execute_trajectory' action
[ INFO] [1560872557.832098937]: Task 'initRos' completed
[ INFO] [1560872557.840900543]: Loading robot model 'ur5_demo'...
[ INFO] [1560872557.840994668]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ WARN] [1560872557.900186396]: Kinematics solver doesn't support #attempts anymore, but only a timeout.
Please remove the parameter '/robot_description_kinematics/manipulator/kinematics_solver_attempts' from your configuration.
[ INFO] [1560872557.922911458]: Loading robot model 'ur5_demo'...
[ INFO] [1560872557.922934313]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ WARN] [1560872557.931521323]: UR5 Descartes Robot Model initialized
[ INFO] [1560872557.931549121]: Descartes Robot Model initialized
[ INFO] [1560872557.931591063]: Descartes Dense Planner initialized
[ INFO] [1560872557.931601840]: Task 'initDescartes' completed
[ INFO] [1560872557.935291554]: Loading robot model 'ur5_demo'...
[ INFO] [1560872557.935329030]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1560872558.923263782]: Ready to take commands for planning group manipulator.
[ INFO] [1560872560.796385655]: Robot reached home position
[ INFO] [1560872560.796522466]: Task 'moveHome' completed
[ INFO] [1560872560.829341247]: Not here1
[ INFO] [1560872560.829453393]: Not here2
plan_and_run_node: /usr/include/eigen3/Eigen/src/Core/CommaInitializer.h:120: XprType& Eigen::CommaInitializer<MatrixType>::finished() [with XprType = Eigen::Matrix<double, 4, 4>]: Assertion `((m_row+m_currentBlockRows) == m_xpr.rows() || m_xpr.cols() == 0) && m_col == m_xpr.cols() && "Too few coefficients passed to comma initializer (operator<<)"' failed.
[plan_and_run_node-1] process has died [pid 20061, exit code -6, cmd /home/ros5/descartes_ws/devel/lib/plan_and_run/plan_and_run_node __name:=plan_and_run_node __log:=/home/ros5/.ros/log/8ab3e7f0-91df-11e9-a030-d43b04c1c3b6/plan_and_run_node-1.log].
log file: /home/ros5/.ros/log/8ab3e7f0-91df-11e9-a030-d43b04c1c3b6/plan_and_run_node-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

jdlangs, 2019-06-24:
For anyone who wants to address this, the error occurs [here](https://github.com/ros-industrial/industrial_training/blob/245438735fb353d8e027378622278bb791d85e9b/exercises/Descartes_Planning_and_Execution/template_ws/src/plan_and_run/src/demo_application.cpp#L205). Eigen is not happy that a 4x4 matrix (representing a homogeneous transform) is being initialized only with 9 values (representing the 3x3 rotation component).

IsaacPiersall, 2019-06-24:
I was able to initialize it as follows, however I assumed the translation component was zero. I do not know if that is a correct assumption.

      rot.matrix() << unit_x(0),unit_y(0),unit_z(0),0
                              ,unit_x(1),unit_y(1),unit_z(1),0
                              ,unit_x(2),unit_y(2),unit_z(2),0
                              ,0,0,0,1;

ngwk1, 2019-07-02:
#280 Fixed the error in demo_application.cpp by changing the matrix for `rot` from 3x3 to 4x4. Please review the changes.

closed 2019-07-02 by marip8