ros-industrial/industrial_training #271: [Perception-Driven_Manipulation] Controller is taking too long to execute trajectory

ngwk1, 2019-07-02:
Controller timed out when running ur5_pick_and_place.launch

**Summary**
[ INFO] [1562032443.559629333]: Execution request received
[ERROR] [1562032451.237170559]: Controller is taking too long to execute trajectory (the expected upper bound for the trajectory execution was 7.557642 seconds). Stopping trajectory.
[ INFO] [1562032451.245451683]: Cancelling execution for 
[ INFO] [1562032451.271039431]: Completed trajectory execution with status TIMED_OUT ...
[ INFO] [1562032451.271208228]: Execution completed: TIMED_OUT
[ INFO] [1562032451.271927874]: Controller  successfully finished
[ INFO] [1562032451.351395402]: ABORTED: Timeout reached
[ERROR] [1562032451.351480201]: Place Move 0 Failed
[ INFO] [1562032451.612303283]: Gripper opened
REQUIRED process [pick_and_place_node-1] has died!
process has died [pid 3033, exit code 1, cmd /home/melodic/p2/devel/lib/collision_avoidance_pick_and_place/pick_and_place_node __name:=pick_and_place_node __log:=/home/melodic/.ros/log/03938b62-9c6c-11e9-ba9b-08002774a5d5/pick_and_place_node-1.log].
log file: /home/melodic/.ros/log/03938b62-9c6c-11e9-ba9b-08002774a5d5/pick_and_place_node-1*.log
Initiating shutdown!

[move_group-2] killing on exit
[pick_and_place_node-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

ngwk1, 2019-07-02:
#

dejaniraai, 2019-07-02:
#272 Pull request to fix this issue. Please review.

dejaniraai, 2019-07-02:
#276 Pull request to fix this issue.

closed 2019-07-02 by marip8