ros-industrial/industrial_training #278: [Demo 2] Reference to header files that do not exist in Melodic

dejaniraai, 2019-07-02:
Reference to headers:
`#include <moveit/kdl_kinematics_plugin/chainiksolver_pos_nr_jl_mimic.hpp> 
#include <moveit/kdl_kinematics_plugin/chainiksolver_vel_pinv_mimic.hpp>`

This was corrected for exercise 4.1 (Descartes), but it is present in the Demo2 template and solution folders.

Derrickakq, 2019-07-02:
#281 Fixed header files by commenting them out as it does not exist on ros-melodic

jrgnicho, 2019-07-02:
fixed by #281

JeremyZoss, 2019-07-02:
Fixed in #281, which is now merged.  Closing this issue.

marip8, 2019-07-13:
This issue also exists in Exercise 5.0; will make a PR soon to fix this

gavanderhoorn, 2019-07-13:
Tracking in #292.

gavanderhoorn, 2019-07-13:
@marip8: I'm closing this again as the problem with ex5 is tracked in #292 and it's best not to 'append' problems to issues already closed.

closed 2019-07-13 by gavanderhoorn