ros-industrial/industrial_training #282: Configure repository to require PR reviews

gavanderhoorn, 2019-07-02:
Just noticed this repository does not enfore PR reviews before merging.

I suggest we setup a policy and -- if available -- also require CI to pass.

dejaniraai, 2019-07-02:
Yes, this is a good idea

gavanderhoorn, 2019-07-02:
It's up to the maintainers typically.

I'll setup a temporary policy and let @ros-industrial/industrial-training-push decide later.

gavanderhoorn, 2019-07-02:
Both `kinetic` and `melodic` have a temporary policy setup.

marip8, 2019-09-30:
@gavanderhoorn I just updated the `kinetic` and `melodic` branches to require passing CI before merge. I'm not sure it's necessary to update earlier distribution branches in a similar way. It seems like this is enough to close this issue. Any thoughts?

gavanderhoorn, 2019-09-30:
I believe this is OK. Indigo and earlier are deprecated, so the current configuration would seem sufficient.

closed 2019-09-30 by marip8