ros-industrial/industrial_training #293: [Demo 1] Perception Driven Manipulation - rosdep errors

IsaacPiersall, 2019-07-16:
Ubuntu 18.04.2 LTS
Ros Melodic

 I started going through the Demo 1 - Perception Driven Manipulation, and when I try and run  the following:

    rosdep install --from-paths . --ignore-src -y

I get the error below:

    ERROR: the following packages/stacks could not have their rosdep keys resolved
    to system dependencies:
    robot_io: Cannot locate rosdep definition for [ur_msgs]

As well when I try to build, I get:

    Errors     << robot_io:cmake /home/ros5/git_workspace/perception_driven_ws/logs/robot_io/build.cmake.001.log
    CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
      Could not find a package configuration file provided by "ur_msgs" with any
      of the following names:

        ur_msgsConfig.cmake
        ur_msgs-config.cmake

      Add the installation prefix of "ur_msgs" to CMAKE_PREFIX_PATH or set
      "ur_msgs_DIR" to a directory containing one of the above files.  If
      "ur_msgs" provides a separate development package or SDK, be sure it has
      been installed.
    Call Stack (most recent call first):
      CMakeLists.txt:5 (find_package)

    cd /home/ros5/git_workspace/perception_driven_ws/build/robot_io; catkin build --get-env robot_io | catkin env -si  /usr/bin/cmake /home/ros5/git_workspace/perception_driven_ws/src/robot_io --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/ros5/git_workspace/perception_driven_ws/devel/.private/robot_io -DCMAKE_INSTALL_PREFIX=/home/ros5/git_workspace/perception_driven_ws/install; cd -

gavanderhoorn, 2019-07-17:
This is caused by the fact that `ur_msg` has not been released into Melodic and it's also not present in the workspace (`--ignore-src`).

closed 2019-09-30 by marip8

marip8, 2019-09-30:
Addressed by #294