ros-industrial/industrial_training #305: Demo 3  catkin build faild

kahmed11, 2020-01-30:
Hello everyone , i am working on that tutorial (industrial training )and its really amazing  and helpfull and everything was going great till i reach demo 3. i actually  wasnt abl to try anything there as am not even abl to  build the workspace.
i tried every possible way that i know to solve that problem but i couldnt know why or how to solve it, so if anyone know the reason or faced the same problem thats gonna be really helpfull and thank you in advance  
when i try catkin build it gives me alot of warning and it end with that error  

Failed     << tesseract_ros:make                       [ Exited with code 2 ]  
Failed    <<< tesseract_ros                            [ 3.9 seconds ]         
Abandoned <<< tesseract_monitoring                     [ Unrelated job failed ]
Abandoned <<< tesseract_rviz                           [ Unrelated job failed ]
Abandoned <<< trajopt                                  [ Unrelated job failed ]
Abandoned <<< tesseract_planning                       [ Unrelated job failed ]
Abandoned <<< pick_and_place                           [ Unrelated job failed ]
Abandoned <<< trajopt_examples                         [ Unrelated job failed ]
[build] Summary: 22 of 29 packages succeeded.                                  
[build]   Ignored:   2 packages were skipped or are blacklisted.               
[build]   Warnings:  6 packages succeeded with warnings.                       
[build]   Abandoned: 6 packages were abandoned.                                
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 4 minutes and 15.9 seconds total.                             
[build] Note: Workspace packages have changed, please re-source setup files to use them.

Levi-Armstrong, 2020-01-30:
I am not sure what the issues is but could you post the CMake errors?

kahmed11, 2020-03-02:
first thing i really appreciate the fast answer from you and really sorry  not being able to rewrite was really out of my hand and my ability 
first thing that i noted that when i do catkin  build it really takes too much time  (around 4 mins ) i dont know if that is normal or abnormal  
there is alot of warnings that appears like  that one  :

_______________________________________________________________________________
Warnings   << bullet3_ros:cmake /home/ros_admin/March_ws/logs/bullet3_ros/build.cmake.000.log
Linux
OPENGL FOUND
/usr/lib/x86_64-linux-gnu/libGLU.so/usr/lib/x86_64-linux-gnu/libGL.so
cd /home/ros_admin/March_ws/build/bullet3_ros; catkin build --get-env bullet3_ros | catkin env -si  /usr/bin/cmake /home/ros_admin/March_ws/src/tesseract/tesseract_ext/bullet3_ros --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/ros_admin/March_ws/devel/.private/bullet3_ros -DCMAKE_INSTALL_PREFIX=/home/ros_admin/March_ws/install; cd -
...............................................................................

but there is only one error that come at the end  and its says  

Errors     << tesseract_ros:make /home/ros_admin/March_ws/logs/tesseract_ros/build.make.000.log
In file included from /home/ros_admin/March_ws/src/tesseract/tesseract_ros/src/kdl/kdl_chain_kin.cpp:36:0:
/home/ros_admin/March_ws/src/tesseract/tesseract_ros/include/tesseract_ros/kdl/kdl_utils.h:33:23: fatal error: Eigen/Eigen: No such file or directory
In file included from /home/ros_admin/March_ws/src/tesseract/tesseract_ros/src/kdl/kdl_joint_kin.cpp:36:0:
/home/ros_admin/March_ws/src/tesseract/tesseract_ros/include/tesseract_ros/kdl/kdl_utils.h:33:23: fatal error: Eigen/Eigen: No such file or directory
compilation terminated.
compilation terminated.
make[2]: *** [CMakeFiles/tesseract_ros_kdl.dir/src/kdl/kdl_chain_kin.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/tesseract_ros_kdl.dir/src/kdl/kdl_joint_kin.cpp.o] Error 1
In file included from /home/ros_admin/March_ws/src/tesseract/tesseract_ros/src/kdl/kdl_env.cpp:56:0:
/home/ros_admin/March_ws/src/tesseract/tesseract_ros/include/tesseract_ros/kdl/kdl_utils.h:33:23: fatal error: Eigen/Eigen: No such file or directory
compilation terminated.
make[2]: *** [CMakeFiles/tesseract_ros_kdl.dir/src/kdl/kdl_env.cpp.o] Error 1
make[1]: *** [CMakeFiles/tesseract_ros_kdl.dir/all] Error 2
make: *** [all] Error 2
cd /home/ros_admin/March_ws/build/tesseract_ros; catkin build --get-env tesseract_ros | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................

really thx in advance