ros-industrial/industrial_training #318: Update Training Material Request

ngwk1, 2020-07-07:
Current presentation slides are still fixed on `ROS-KINETIC` version. 

### [Session 0](https://github.com/ros-industrial/industrial_training/blob/melodic/gh_pages/_downloads/slides/ROS-I%20Basic%20Developers%20Training%20-%20Session%200.pdf) 
- [ ] Update GUI outlook in the presentation slides  (Slide 4 - 10)
- [ ] Introduce an additional command to allow user to `right-click` and add `New Document` in any folders. Fixed by
```
touch ~/Templates/"Untitled Document"
```
### [Session 1](https://github.com/ros-industrial/industrial_training/blob/melodic/gh_pages/_downloads/slides/ROS-I%20Basic%20Developers%20Training%20-%20Session%201.pdf)
- [ ] Slide 31: Still under Kinetic (Slide 31)

sachinkum0009, 2020-07-07:
Update GUI outlook in the presentation slides (Slide 4 - 10)
These changes have been done

sachinkum0009, 2020-07-07:
BTW, we can not right-click and add a new document in the files. Only new folder command is available when right-clicked.

cbw36, 2020-07-07:
I'd be happy to wrap up this issue. Can it be assigned to me? Is there anything that needs to be done other than change Session 1 slide 31 to melodic? I will also look for other references to kinetic and change it.

gavanderhoorn, 2021-09-29:
@marip8: should this be closed seeing as #321 was merged?

marip8, 2021-09-29:
Yes, I believe this can be closed

closed 2021-09-29 by marip8