ros-industrial/industrial_training #322: [Demo 1] Perception-driven manipulation "Robot model parameter not found"

max-sn, 2020-07-08:
Setup:
* Native Ubuntu 18.04 LTS install
* `.check_training_config.bash` result is all [OK]

I checked both the `template_ws` and the `solution_ws`, and both build fine, but I get the following (fatal) error(s) when directly after starting `roslaunch collision_avoidance_pick_and_place ur5_pick_and_place.launch`:

```
[ INFO] [1594218924.570461140]: Parameters successfully read
[ERROR] [1594218924.572215833]: Robot model parameter not found! Did you remap 'robot_description'?
[ERROR] [1594218924.577955019]: Robot model parameter not found! Did you remap 'robot_description'?
[FATAL] [1594218924.578414487]: Unable to construct robot model. Please make sure all needed information is on the parameter server.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Unable to construct robot model. Please make sure all needed information is on the parameter server.
[ERROR] [1594218924.601495686]: Robot model not loaded
[ERROR] [1594218924.610968811]: Planning scene not configured
```

max-sn, 2020-07-08:
I found that changing https://github.com/ros-industrial/industrial_training/blob/ddfcfa346def75010de087a7b20c61dd41d326e6/exercises/Perception-Driven_Manipulation/template_ws/src/ur5_collision_avoidance_moveit_config/launch/move_group.launch#L3
into
```xml
<include file="$(find ur5_collision_avoidance_moveit_config)/launch/planning_context.launch" >
    <arg name="load_robot_description" value="true" />
</include>
```
makes me able to run the launchfile without error, but I am unsure whether that is the right solution. When I apply this there's still no RViz screen opening.

marip8, 2020-07-09:
It's not entirely clear, but there are two launch files that need to be run for this exercise: a setup launch file (`collision_avoidance_pick_and_place/launch/ur5_setup.launch`) and an application launch file (`collision_avoidance_pick_and_place/launch/ur5_pick_and_place.launch`). The setup launch file sets the `robot_description` parameter among several other things. The application launch file only runs the pick and place node that was developed as a part of this exercise.

The solution you suggested fixes your issue, but the demo will not work properly without also launching the setup launch file. I think the right solution in this case is to amend the training documentation to say that the setup launch file should be run first before launching the application.

```diff
+ roslaunch collision_avoidance_pick_and_place ur5_setup.launch
roslaunch collision_avoidance_pick_and_place ur5_pick_and_place.launch
```

This should be added to the documentation in the sections Demo 1.3, 1.5, 1.6, 1.7, 1.8, 1.9, 1.10

max-sn, 2020-07-10:
Ah I missed that one. Would it also be interesting to convert the Markdown files to reStructuredText? Since you're working with Sphinx, it might prove to be easier to keep up to date (e.g. by using `.. literalinclude::` for including code). I could add the additional launch file instruction too.

marip8, 2020-07-10:
I think reStructuredText could be beneficial for a lot of these files; we've used it on some of the newer pages, but we haven't had the chance to convert the older pages. I would be happy to review and merge a PR if you want to convert this page to RST. Otherwise a simple PR with the change described above would be much appreciated.

PR #325 (2020-07-14): Converted demo 1 instructions to reStructuredText
Includes:
* The instructions for demo 1 are converted to reStructuredText
* Instructions are added to fix #322. 
* For my build process I also added support for Sphinx versions >=1.4, this shouldn't break anything for lower versions.
closed 2020-07-14 by marip8