ros-industrial/industrial_training #331: Tutorial Part 3.2 not up-to-date.

JoeGre666, 2020-11-09:
Hey, I'm currently trying to do ROS Industrial (Melodic) Training Exercises 3.2. I did everything as in the instructions, but always got errors. According to the error message there is no buffer.h / transform_listener.h in the package tf2, so I looked it up and found the matching header files in the tf2_ros package (look at instructions).

[image]

So I tried including those packages, still errors. After a few more tries I looked in the exercise-files and found out that I was right about the tf2_ros-package, but additionally the <tf2_geometry_msgs/tf2_geometry_msgs.h> header-file is missing (screenshot is from exercise files - for reference). This is not mentioned in the instructions at all.

[image]

Since I was a bit frustrated, I suggest to maybe adjust this? :) 
Apart from that - great tutorial.
Kind regards,
Johannes

Baschalp, 2021-01-30:
Thank you for the suggestions, this is really helpful. Otherwise I was going to submit a similar content regarding the changes.

marip8, 2021-02-01:
Thanks for catching this issue. It seems like we actually changed the solution to the exercise [here](https://github.com/ros-industrial/industrial_training/blob/2f9a48969ee8da2e4f4d18cca1a408f2bc6b32ba/exercises/3.2/src/myworkcell_core/src/vision_node.cpp#L8-L10) to be correct, but we didn't catch the header references in the exercise documentation.