ros-industrial/industrial_training #341: Any Plan with motion planning with C++ ROS Kinetic

kappa95, 2021-03-23:
Hi I am following your tutorials and everything worked till Exercise 4.0 for motion planning with C++. It doesn't find any plans for any possible motion **providing Pose or PoseStamped messages** Instead If i move joints positions works.  Some suggestion? I missed something?

marip8, 2021-03-23:
I would suggest checking your code against the files for the solution to the exercise [here](https://github.com/ros-industrial/industrial_training/tree/677de0b6067a56696e5c13bee9d5310c24e56c0c/exercises/4.0/ros1/src/myworkcell_core/src). My guess is that there is a small discrepancy between your code and the working solution that is causing this issue. If so, we could look into the wording of exercise 4.0 to make the instructions more clear

kappa95, 2021-03-23:
Find the mistake: In the `vision_node.cpp` everything was with the library: `tf` instead of `tf2_ros`.
Modifying as in the file you linked everything worked! Thank you so much!

closed 2021-03-23 by kappa95