ros-industrial/industrial_training #373: Broken link for Pre-requisites

robogeekcanada, 2021-11-26:
Not sure if raising an issue is the right way to point this out for observations on the training material.

Under the Pre-requisites section, the link for Bruce Eckle's Thinking in C++ is broken 

```
Prerequisites
-------------

C++
~~~~~~~~~~~~~

.. toctree::
   :maxdepth: 1

   MIT Introduction to C++ <http://ocw.mit.edu/courses/electrical-engineering-and-computer-science/6-096-introduction-to-c-january-iap-2011/assignments/>
   Bruce Eckel Thinking in C++ <http://mindview.net/Books/TICPP/ThinkingInCPP2e.html>
```

Suggested new link: https://web.mit.edu/merolish/ticpp/TicV2.html

mariavergo, 2022-07-06:
Update to new link: http://chuckallison.com/TICV2/html/TicV2.html

hanifz-dr1, 2022-07-06:
Fixed in PR #377

commit c03c6c6a (2022-07-07): Merge pull request #377 from hanifz-dr1/fix-eckel_link_update

Replaced Bruce Eckel's Thinking in C++ broken link with new links. Resolves #373
closed 2022-07-07 by robinsonmm