ros-industrial/industrial_training #395: UR5 not displaying in Rviz - Training 3.1 Workcell XACRO

lucasbrodo, 2023-05-29:
Hello,

I'm following in order the training tutorials but I'm encountering a problem in tutorial 3.1 - Workcell XACRO. The UR5 robot is not represented even if everything works with TF after I run the final command `ros2 launch myworkcell_support urdf.launch.py` (see picture below).

[image]

Did I miss something ? 

Thank you very much for your help and for all the great training material.

Best,

Lucas

marip8, 2023-05-30:
Can you post the terminal output for this issue?

lucasbrodo, 2023-05-30:
Hi @marip8,

[image]

Since I have a german locale setting, I tried the recommended `export LC_NUMERIC="en_US.UTF-8"` (https://github.com/ros-industrial/universal_robot/issues/374#issuecomment-411025881) but it does not seem to work.

marip8, 2023-05-30:
Interesting; usually when the mesh models don't show up, it's because the package URI's can't be resolved (sometimes because the repo wasn't sourced correctly), but that doesn't seem to be the case here. Have you set your `ROS_DOMAIN_ID` or `ROS_LOCALHOST_ONLY` environment variables? Sometimes I see weird behavior like this when multiple people are running ROS2 systems simultaneously on the same network. If you don't want to set those variables, you could also disconnect from LAN/WAN and try running this launch file again

lucasbrodo, 2023-05-30:
Hi @marip8, I have tried setting the `ROS_DOMAIN_ID` and `ROS_LOCALHOST_ONLY` but it does not change the output. Neither disconnecting the LAN/WAN. 
Do you have any other ideas ?
Thank you for your help