ros-industrial/industrial_training #404: Sources of pdf slides for translation

gautz, 2023-09-21:
I'd like to translate the pdf slides in French. Are these Open Source ? If yes, could you add the (Latex?) sources ?

marip8, 2023-09-21:
Yes, the slides are open-source. I'll see if I can track down the sources. They're probably .pptx format, so I may have to send them to you directly based on how large the files are. FWIW, Google Translate can translate PDF documents. [Here is one of the files I tried](https://github.com/ros-industrial/industrial_training/files/12691123/ROS-I.Advanced.Developers.Training.-.Building.A.Perception.Pipeline.2.pdf). If the translation is any good, this might be a reasonable approach. I think Adobe can do translation as well, but I'm not sure about its quality either

gautz, 2023-10-08:
I've sent you an email with an upload link if you can find the sources.