ros-industrial/industrial_training #408: Industrial Training Demo 1.2 Fails to compile

BHolmes92, 2023-10-31:
Currently using the humble vm to work through the first demo. Colcon build fails due to an error in set_attached_object.cpp

`/perception_driven_ws/src/pick_and_place_application/src/tasks/set_attached_object.cpp:34:30: error: no matching function for call to 'moveit::core::RobotState::attachBody(std::string&, std::vector<std::shared_ptr<const shapes::Shape>&, EigenSTL::vectorIsometry3d&, std::vector<std::__cxx11::basic_strin<char>&, std::string&)'`

It looks like const trajectory_msgs::JointTrajectory is missing?

reidchristopher, 2023-10-31:
This was a bug that got fixed shortly after the VM was published, so pulling the latest changes for the template and then copying into your workspace should fix it (see #400)

BHolmes92, 2023-11-01:
perfect that did the trick, would it make sense to add a note to the doc until the vm is updated?

closed 2023-11-01 by BHolmes92

reidchristopher, 2023-11-01:
There's already instructions to pull down the latest changes to your VM on the [PC setup page](https://industrial-training-master.readthedocs.io/en/humble/_source/setup/PC-Setup---ROS.html), it can just be easy to overlook