ros-industrial/industrial_training #416: Clearly communicate ROS-I's status in ROS2

b-adkins, 2024-06-25:
# Background
The great ROS to ROS2 migration has been confusing across the board. It's often difficult to untangle which tutorials, documentation, repos, and packages are used in each version, especially as some ROS2 packages are ports, but others are wrappers/adapters that include the ROS1 library.

# Problem Statement
I found myself confused for an hour by what should be simple questions:

- Does ROS Industrial support ROS2?
- Where is the documentation for ROS-I in ROS2?
- What are the core packages for ROS2?

## ROS2 Support
I had to read between the lines to figure this out:

- There are training materials for Foxy (a ROS2 distro): https://github.com/ros-industrial/ros2_i_training. Last update, 8 months ago. Appears to be deprecated, as there is a `foxy` branch in:
- There is a version of training docs with ROS2 branches: https://github.com/ros-industrial/industrial_training. Last update, 2 months ago. Hosted at https://industrial-training-master.readthedocs.io/en/humble/.
    - Buried in the "Archive" is mention of a [ROS1-to-ROS2 bridge](https://industrial-training-master.readthedocs.io/en/humble/_source/_archives/ROS1-ROS2-bridge.html). There is a long history of running different ROS versions for a single robot, but that's not something I would expect a beginner to know.

- industrial_core has no branches for ROS2 distros, but a branch `ros2_msgs_only` which doesn't elaborate further.
- Tucked away discussion in an issue thread: https://github.com/ros-industrial/industrial_core/issues/253

So... barely?

## ROS-I Docs for ROS 2
- Not the [ROS1 wiki page](https://wiki.ros.org/Industrial), but that's where [rosindustrial.org](https://rosindustrial.org/) sends you.
- This appears to be it. [Industrial Training](https://industrial-training-master.readthedocs.io). API reference materials do not appear to be available. They are hard to build yourself without knowing which packages are available in ROS2.

## Core packages
The core repo from ROS1 appears to not be ported much. The interfaces must be checked out from this branch and built:

https://github.com/ros-industrial/industrial_core/tree/ros2_msgs_only

Many ROS-I clients seem to not be ported.

# Proposed Changes
- Add these answers to the FAQ on rosindustrial.org
- Landing slide of this documentation should mention:
    - The most recent status of ROS2. (Interface must be built from source, [many packages have yet to be ported/redesigned](https://github.com/ros-industrial/industrial_core/issues/253).)
    - Concurrent ROS1 and ROS2 environments. It's a venerable solution in the ROS world to have two ROS environments running at the same time, but newcomers may not realize it. Until the ROS2 migration finishes in several years (decades?) this approach will have to be on the table for devs who need something now. See [how to bridge ROS1 to ROS2](https://industrial-training-master.readthedocs.io/en/humble/_source/_archives/ROS1-ROS2-bridge.html).
- Discuss what can be done to make the documentation environment clearer.

robinsonmm, 2024-06-26:
Thanks for taking the time to post this issue. You raise a number of great points that we need to provide clarity on relative to ROS 2 and some of the legacy tools that were commonly used in ROS 1. We will be discussing this among the tech contributors and come up with a reasonable path forward to drive more clarity.

robinsonmm, 2024-11-22:
Update on our end. We have a revised FAQ in review. I understand this seems to have drug on, but there are a few stakeholders in the loop. We are also adding all GitHub orgs to the ros-i.org site and noting in GitHub org level readmes back to ros-i to help. 

We had the bridge exercise as an advanced topic. We now have additional bridge resources, including a clever Noetic to Foxy container to allow for Humble + distros of ROS 2, but could use some additional documentation to support. 

I will add more additional detail here as we push more changes visible, and have a more firm plan on what to do about the rosindustrial wiki. 

Thanks.

MR